sinara-hw/RFSOC-AMC #1: [RFC] RFSoCs from Xilinx

gkasprow, 2018-12-21:
I got involved in new project at the WUT. 
We will build wideband MIMO SDR. Probably on the new RFSoC that:
- are pin compatible with ZU25
- offer 8x 10G DACs and 8x 5G ADCs

You won't find any info with Mr Google. I got this news from my colleague. 
For the WUT project existing 6G DAC with DUC and 4G ADC with DDC would probably be enough, but I want to play with newest SOCs.
For what applications would you use such beast? Would such upgrade for Sayma ecosystem be useful for ion trappers?
My colleagues from NBI would love to use it for superconducting qubit readout.
I still don't know what the ADC-DAC latency is. They need <1us together with demodulation and modulation algorithm. Both ADC and DAC use AXI stream.
I plan to buy ZCU111 and simply measure it.
The configuration for WUT project is minimal. I want to use 1517 package an hook up as many DDR4 channels as possible for all AWG applications.
Is there any benefit of using Sayma AMC in such configuraiton? 
Initial idea is to make it as RTM and place SFP/QSFP, clock recovery, MMC, supply on AMC side.

camillehyrien, 2019-02-25:
> I saw it. But it has so many variables like inter-tile alignment and various synchronisation features.
> I will try to simulate the design with Vivado simulator where I can enter analog values and observe analog outputs.

Do you find another way to run simulation? (for example to see spectrum ? ) 

I have another question, do you find a way to synchronize your mesurement equipment and the ZCU111 dev board with LMK2594? Do you need to configure a MTS mode, or non-MTS mode? In which way you connected it? I tried several times but It failed.

gkasprow, 2019-02-25:
I'm still waiting for ZCU111 devkit. I didn't touch it for some time since it is a low priority.

gkasprow, 2019-09-08:
We measured the latency:
ADC - DMA - DAC. And it is roughly 160ns with interpolation and decimation and 109ns without.
Some results are [here](https://gitlab.com/wut_ise/pados/issues/8)

camillehyrien, 2019-09-09:
> We measured the latency:
> ADC - DMA - DAC. And it is roughly 160ns with interpolation and decimation and 109ns without.
> Some results are [here](https://gitlab.com/wut_ise/pados/issues/8)

Thank you for this information, I am not able to see the link : https://gitlab.com/wut_ise/pados/issues/8. May you post the result in a pdf format? Thank you

gkasprow, 2019-09-09:
This result (160 ns delay) is when using 2x interpolation and 2x decimation on DAC and ADC respectively. 4,096 GSPS.
[image]
[image]

109 ns is without interpolation and decimation and 4,096 GSPS, however you cannot set this configuration from GUI as GUI prevents DAC AXI stream clock from being higher than 500 MHz. I did this from PYNQ level.
[image]

gkasprow, 2019-09-09:
__
[image]

gkasprow, 2019-09-09:
The cable lengths were compensated

gkasprow, 2021-10-04:
PCB progress update
[image]
[image]
[image]
Mezzanine
[image]
[image]

dtcallcock, 2021-10-04:
FYI, there is now an open-source superconducting qubit project based on an RFSoC eval board.

[https://arxiv.org/abs/2110.00557](https://arxiv.org/abs/2110.00557)
[https://github.com/openquantumhardware/qick](https://github.com/openquantumhardware/qick)

Sandia have a similar thing aimed at ion traps too.  Not sure if that will be opened up.

[https://www.osti.gov/doecode/biblio/46631](https://www.osti.gov/doecode/biblio/46631)

gkasprow, 2021-10-04:
Yes, I know. I wrote to David already :)