sinara-hw/RFSOC-AMC #2: Design choices

gkasprow, 2018-12-21:
So far several aspects were decided:
- AMC form factor
- 2 AFE modules a'la Sayma
- RTM compatible with [RTM-SFP-QSFP](https://github.com/gkasprow/RTM_4SFP_3QSFP)
- AMC FatPipe1 support for DRTIO and PCIe
- clock racovery same as on Sayma (WR-type) + Xilinx devkit-style ADC/DAC clock generation
- 2x 64bit PL-DDR4
- 1x64bit PS-DDR4
- power supplies based on uModules like on Sayma RTM
- small FPGA for AFE control like on Sayma RTM

tprzywoz, 2019-03-18:
@gkasprow  I've just published block schematics. Please check if connections on top schematic meet the requirements.

drewrisinger, 2019-12-04:
Have there been any updates since March?

gkasprow, 2019-12-04:
Yes, the schematics are completed. We also got funding so in January we will focus on routing.