sinara-hw/RFSOC-AMC #3: AFE Connector

tprzywoz, 2020-04-28:
Will the connectors used on devkit be sufficient or do we want something with higher bandwidth?
https://www.samtec.com/products/lpaf-40-03.0-s-08-2-k-tr

gkasprow, 2020-04-28:
We can use:
-  either the same that XIlinx is using on their devkit + additional IO and LVDS lines
-  or copy solution from Sayma RTM

gkasprow, 2020-04-28:
Isn't the footprint (pitch) for both connector types compatible?

tprzywoz, 2020-04-28:
> Isn't the footprint (pitch) for both connector types compatible?

They have only different NPTH (2) and LPAF (Xilinx devkit one) has max 8 rows (not 10 like ASP on Sayma RTM). LPAF has slightly better high speed performance (based on a 3dB insertion loss) - 18.5 GHz vs 18 GHz of ASP.

tprzywoz, 2020-10-09:
After discussion with @gkasprow we are considering changing "the AFE standard" to Xilinx ZCU111 devkit one. In the current version there is copied Sayma solution:

2 x SAMTEC_ASP-134488-01 (40 x 10) connectors. Each has connections for:
- 4 x ADC and DAC channels
- 16 x FPGA IO (bank 3.3V)
- 8 x LVDS
- GT Link
- 2 x I2C
- Power rails: +/- 12V ; +/-6V ; +3V +2.5V

Potentiall new "Devkit standard":
2 x LPAF-40-030-S-08-2-K-TR (40 x 8)
**"ADC connector":**

- 8 x ADC channels (and 8 VCM)
- 20 x FPGA IO (HD bank 1.8V)
- 4 x ADCLK_IN (diff) - alternative clock source variant for each ADC CLK input
- I2C
- Power rails +3V and 1.8V

**"DAC connector":**

- 8 x DAC channels
- 20 x FPGA IO (HD bank 1.8V)
- 2 x DACCLK_IN alternative clock source variant for each DAC CLK input
- Power rails +12V, +3V, +2.5V (DAC_AVTT),+1.8V 

First of all, that change will reduce the number of power rails. Currently, it makes a problem with component placement:
[image]

Thanks to lack of +/-12V +/- 6V power regulators and with AFE depth reduction to 76.5 mm (like FMC) we can fit all power regulators without adding extra PCB modules (as a solution to increase PCB area). Other advantages are better signal integrity (thanks to shorter traces and better isolation of analog part of PCB) and well-known connectors dedicated to RFMC (instead of FMC).

Disadvantages:

- RFSoC will accept only double width AFE boards (12V only on ''DAC connector'', I2C only on ''ADC connector")
- We won't be able to use AFE boards designed for ZCU111 devkit - the mechanical standard is too wide to fit on AMC:
[image]

gkasprow, 2020-10-09:
But we will be able to plug the Xilinx mezzanines to our board and our mezzanines to XIlinx devkit because connector distance would be the same. This is useful test/characterization purposes. Of course we won't be able to insert them to the AMC slot, but it's not a big issue.

gkasprow, 2020-10-09:
Moreover, we didn't characterize the Sayma connectors (FMC ones) for operation in such a high-frequency range. It's safer to stick with Xilinx devkit layout and connectors because it was already proven to work.