sinara-hw/RFSOC-AMC #45: Power budget

filipswit, 2020-10-27:
Use LTpowerCAD to make power budget. With nice drawing will be easier to verify if there aren't any more mistakes.

tprzywoz, 2020-11-10:
I added to repo rfsoc.ltp3 file. I couldn't find information about RFMC power limits for Xilinx Eval but based on ZCU111:

- 2.5V (DAC_VTT) was designed for max 500mA (including SoC and RFMC)
- 1.8V (VCC1V8) was designed for max 2A. We have only additional Artix-7 and 4A current limit - should be enough.
- 3.3V it's really hard to estimate ZCU power limit for this rail. In current RFSoC_AMC design, current limit is 2,7A (all IC worst case). It can be increased to 6,7A by merging P1V13 and MGT1V2.

filipswit, 2020-11-21:
Good job! 

AMC module (+eventual RTM) shouldn't draw more than 80W, now it's almost 100W.
Same with 3V3_MP, it should not exceed 150 mA

But I don't think you can do anything with it.

closed 2021-06-27 by filipswit