sinara-hw/RFSOC-AMC #53: Components availability

kaolpr, 2021-02-22:
After reviewing BOM at we have the following remarks:

* LPC1776FET180,551 has limited availability (not stocked by major distributors) - this will be a problem for prototype series, if keeping the same package maybe LPC1788FET180,551 could be an alternative?
* VX-501-0251-122M88 has 18 week lead time, I hope this **will be** a problem :P, can we have some alternative here?
* XC7A35T-3CSG325E is not available on stock, can we use XC7A50T-3CSG325E? @sbourdeauducq - I assume that speed grade 3 is required?

sbourdeauducq, 2021-02-23:
Yes, SG3 is better generally.

sbourdeauducq, 2021-02-23:
And 7A35T and 7A50T have the same die inside, so it'll work regardless of what Xilinx thinks :)

tprzywoz, 2021-02-24:
We can use CVHD-950-122.880 instead of VX-501-0251-122M88.

filipswit, 2021-05-16:
Heatsink **AAVID** **342947** looks unavailable.

I suggest to use same as on AFCZ: **SKV585811-CU**

tprzywoz, 2021-06-10:
Heatsink changed to SKV585811-CU