sinara-hw/RFSOC-AMC #83: components to close to each other

filipswit, 2021-05-23:
- [ ] IC63 is to close to C936
- [ ] This selector will be hard to change with mounted radiator
[image]

filipswit, 2021-05-29:
- [ ] C373 C372, can be moved to the left, to wouldn't overlap of IC23 courtyard
[image]

filipswit, 2021-05-29:
- [ ] This selector will be hard to change with mounted radiator

[image]

tprzywoz, 2021-06-08:
All done. The first selector moved to BTM:
[image]

closed 2021-06-27 by filipswit