sinara-hw/RFSOC-AMC #97: RFSOC-AMC SI verification

tprzywoz, 2021-06-22:
PL0   DDR4_2666

Data Write
[image]
[image]
[image]
[image]

Data Read
[image]
[image]
[image]

Address
[image]
[image]
[image]
[image]
[image]

Differential Nets
[image]

CLK-DQS Skew
[image]

tprzywoz, 2021-06-22:
PL1 DDR4_2666

Data Write
[image]
[image]
[image]
[image]

Data Read
[image]
[image]
[image]
[image]

Address
[image]
[image]
[image]
[image]
[image]

Differential Nets
[image]

CLK-DQS Skew
[image]

kaolpr, 2022-01-26:
@tprzywoz Have you done simulations for MGT lines?

tprzywoz, 2022-01-27:
Yes, to AMC edge connector with full speed, to RTM connector 12 Gb/s.

gkasprow, 2022-01-27:
Is there a chance that this board could work at 28Gbit/s over RTM ? We can use a higher grade RTM connector.

tprzywoz, 2022-02-02:
Eye diagrams are fine for 25Gbit/s. At 28 Gbit/s there are problems with the last Quad which has mounting options and more vias.

gkasprow, 2022-02-02:
I assume this is without the RTM board.
Anyway, there is an idea to use Samtec Flyover to QSFP cable. From one side we exit close to MGT, from the other side we have a QSFP28 connector. We only need to find a place for Flyover connector, but it is tiny,
We could implement it in new revision,