sinara-hw/RFSOC-AMC #99: status of ARTIQ port to an RFSoC

jbqubit, 2025-01-10:
The thesis by Brad Bondurant (2024) discusses work toward a port of ARTIQ to the ZCU111. From conversations with Brad and, now, reading his thesis I surmise that he got slowed down by porting the Xilinx-provided C libraries to rust. Definitely the right long-term approach but pretty hard as Xilinx code is notoriously brittle. To my eye it looks like he made great progress and the narrative accompanying his code (in the thesis CH 5) is quite helpful.  

https://dukespace.lib.duke.edu/items/6d3950ae-a4ea-4caf-bc8e-2e491eeddbc3

The source is here. 

https://gitlab.com/duke-artiq/zynqmp-rs

As far as I know a functional ARTIQ port to an RFSoC remains nascent. I hear Warsaw (@gkasprow) is also working on a port. I gather M-Labs supported parts of Brad's port (@sbourdeauducq). Anybody know the status of these efforts? Are any of the efforts inclusive of the ZU48DR used in the AMD RFSoC 4x2?

gkasprow, 2025-01-11:
@kaolpr

kaolpr, 2025-01-11:
We have one person working on the subject based on work by Brad. He's in contact with Brad and currently we assume the approach as in case of Zynq7000.

However, I'm curious what were the arguments against  Linux based approach (ARTIQ app on Linux + RT cores). That could prevent us from reverse engineering Xilinx codes.

pjedyk, 2025-03-14:
I have started working on a Zynq UltraScale+ port - specifically for Genesys ZU-5EV, but it should be easily portable to any ZCU.

Currently, I can share a suggestion for a development environment.

https://github.com/pjedyk/artiq-new/tree/wip/boards/genesys_zu-5ev

The firmware part will be stacked Rust on C, so the regular Xilinx/AMD C runtime jumps into Rust after initialising all the CPU and C runtimes.

The gateware part is in Migen. I have written experimental Migen-Vivado integration that generates a Platform class based on the Vivado project.

The environment is designed to keep costly reverse engineering to a minimum. It includes formatting/linting facilities for Python.

I plan to use AMP (maybe with OpenMP integration?). I consider some OS-es (FreeRTOS?) that could use MPU/MMU. I also plan some experiments with FreeRTOS SMP; however, that probably won't be the case.

I also plan to distribute tasks between Cortex-R and Cortex-A cores.

kaolpr, 2025-03-15:
@sbourdeauducq What are your experiences with Not OS used for Nix Servo? Do you think this would be applicable here (boldly assuming  that we accept the use of Linux)?

pjedyk, 2025-04-11:
I have successfully done several experiments with Linux and Zynq US+ on Genesys ZU-5EV and confirm that it is possible and has great potential to use the following architecture:

- Low-level firmware in split-mode AMP on two Cortex-R cores, talks with PL via AXI, with Cortex-A53s via OpenAMP. Written mostly in Rust, compiled to static *.a using cargo. Glue layer in C, compiled using Vitis and linked with Rust's *.a. C Layer provides OpenAMP and Xil HAL without the need to rewrite them (extern "C" in Rust).
- Middle-level firmware as Linux on four Cortex-A53 cores, talks with Cortex-Rs via OpenAMP, PL via AXI, and the rest of the world via Ethernet and UART. We will use Yocto/OE with meta-xilinx, so no petalinux tooling is needed. I will manage not to put any bloatware to avoid annoying the neurotic guys (as I am) - OE with a custom image recipe will help.
- PL side, talks with PSU via AXI and controls physical inputs/outputs.

**I need an answer about what approach you feel more comfortable with (taking into account the design of the future custom PCBs):**

**Approach 1**: Linux boots with a RAM-stored read-write but volatile filesystem. After reboot, you lose all changes to the filesystem. However, you may mount and use non-volatile additional storage (SD cards, USB, etc.). There will also be a small, non-volatile partition for configuration (hostname, net config, etc.). You will never "brick" the device by accident unless you are very talented. You have a well-known state of a device after each reboot.

**Approach 2**: The device boots entirely from storage like an SD card or eMMC. Thus, you have a regular computer and may see previous users' trash files and get upset. You may easily break something (sudo dd, which means "destroy disk"), but you have an unlimited environment.

What are your thoughts?

If no preference, I would choose approach 1.

pjedyk, 2025-04-11:
For me, Linux is not a must. However:

- We have four Cortex-A53 cores with Mali-400 which will run Linux smoothly and additional two Cortex-R5F cores that can do the low-level job. The Zynq SoC is designed this way - it is "by the book" approach.
- Linux provides easy-to-use and understand by everyone: network interfaces configuration and management, Bash console over UART and SSH, network services like mDNS (discovering by hostname.local), ...; it adds value in the case (IMO, of course).
- Well-tuned Yocto/OE build should be much more stable than writing everything that in bare-metal, and much, much faster in development (also includes future ports to different devices).

I believe that once the new approach development environment is stable, the speed of the further development will rocket-out bare-metal grinding. I do not have a prove for that, tough.

Please consider the noteworthy fact that it is not "moving ARITQ to Linux" completely. It is delegating some well-done in Linux functions away from bare-metal, but still keeping the "main", or "novel", code in Rust on 32-bit Cortex-R5. Thus, this shall be named rather functional "balancing" to provide an optimised allocation.

Any comments are welcome; please provide as much point-of-views as possible to increase the chance of good decision.

kaolpr, 2025-04-11:
I think what @pjedyk had in mind here was stability of development.

I agree with arguments for Linux + IMO Linux as a management OS will make getting new developers a bit easier (although still not easy). It is hard to persuade someone to dedicate his time to work on the solution that is unique to a relatively narrow field. Even if this solution is technically impressive.

pjedyk, 2025-04-11:
> I think the kernels should be run on the main cores if possible.

It is possible, however, that my strategy was to first balance Linux on A53 and Rust on R5, and if this is not enough, I will put Rust on two A53s (for example).

We may run Linux only on two A53s and keep the other two for Rust (low-level) using standard Xilinx tooling - no need for reversing anything, just a different configuration.

> Is ARTIQ on Zynq unstable due to driver bugs?

The artiq-zynq repo supports Zynq 7000, not UltraScale+, afaik.

Zynq 7000 port is not trivial (delicately said) to port to Zynq US+.

I suggest an approach that radically simplifies the development. Future programmers should focus on ARTIQ functionality, not reversing HAL, rewriting bootloaders, initialising C runtime, and writing drivers for standard hardware like Ethernet.

---

Please provide information if keeping a closed ecosystem is a strict policy. If not, I recommend trying the proposed architecture and let the community test it.

I agree that "Quantity != quality", but, as I said, well-tuned (it is a keyword: **well-tuned**) Vivado+Migen+Rust+Vitis+Yocto/OE workflow shall be more on the quality side.

gkasprow, 2025-04-11:
@sbourdeauducq, we have only one man working on it and minimal financial resources. Let's make a minimal viable prototype approach first.

pjedyk, 2025-04-11:
This thread started to have traits of a political debate, not an argument-driven technical discussion.

The comment "Try writing a bare metal ... that writes something to the UART ... you'll see it's not that hard" implicitly declares that I have never used registers to handle the hardware, which is incorrect.

Another thing is that even if handling UART is not that hard, handling GPIOs is not that hard, and doing many other things is not that hard... doing it all together may be hard. It is the fallacy of composition.

I expected to:
- gather arguments for Linux
- gather arguments for bare-metal
- discuss and compare
- make decision
- implement

I can write the firmware entirely in Rust from scratch (with interrupt vectors and low-level init). I was writing custom PendSV interrupt handlers on 32-bit ARMs in assembler to switch contexts. I even used MPU on STM32 before FreeRTOS officially added it to its upstream. BUT, every time I had a serious reason to do things that way - I have never done it for "a challenge".

I wanted to use Linux to:

- Take advantage of several drivers and subsystems, particularly storage devices, filesystems, and networking.
- Improve the development and maintenance by implementing some functions (that do not require real-time awareness) under Linux using languages like Python.
- Improve setup from the user's point of view, introduce log gatherings, and store experiment results on storage devices.
- Reduce the effort of porting to future custom PCBs with Zynq US+.

Functions that should be deterministic and real-time aware are still bare metal and PL-side, so we are not resigning from anything.

I am entirely open to solid arguments.